RT-Thread-packages/at_device #44: [esp8266] wifi热点关闭再重新打开后，程序会进入hard fault

z14git, 2019-04-23:
3月8号的[版本](https://github.com/RT-Thread-packages/at_device/tree/d4afe74e87afc7927282f5be4ce320f826755dc9)，在遇到wifi热点突然关闭然后再打开的情况，程序是可以正常运行的。但现在更新到[最新版本](https://github.com/RT-Thread-packages/at_device/tree/a6c434b4f6c99cc3810ab4dd48081b2c8affaf5e)后（rt-thread也更到[最新](https://github.com/RT-Thread/rt-thread/tree/ca7ebe61750b27c4e85f79c29b6596bc799cf016)），当遇到这种情况，程序就会进入hard fault。
不过目前还不确定是rt-thread的问题，还是at_device的问题。
以下是进入hard fault前的记录：
```
[I/at.esp8266] ESP8266 WIFI is disconnect.
[E/onenet.edp] Received error.
[E/sal.skt] not find network interface device by protocol family(45).
[E/sal.skt] SAL socket protocol family input failed, return error -3.
[E/onenet.edp] socket create failed
[I/at.esp8266] ESP8266 WIFI is connected.
[E/at.clnt] execute command (AT+CIPDOMAIN="link.rt-thread.org") timeout (5000 ticks)!
psr: 0x61000000
r00: 0x00000000
r01: 0x6b726f77
r02: 0x200017b0
r03: 0xffffffff
r04: 0x2000206c
r05: 0x20003470
r06: 0x200020a8
r07: 0x00000000
r08: 0x0801e018
r09: 0x20003600
r10: 0x20003470
r11: 0x20003600
r12: 0x08018299
 lr: 0x08009b8d
 pc: 0x6b726f76
hard fault on thread: sys_work

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
onenet    25  suspend 0x00000098 0x00000500    57%   0x00000002 -95
tshell    20  suspend 0x00000138 0x00000800    25%   0x0000000a 000
at_clnt    9  suspend 0x000000ac 0x00000600    20%   0x00000004 000
sys_work  23  running 0x000001ac 0x00000800    27%   0x00000006 -02
tidle0    31  ready   0x00000040 0x00000100    31%   0x00000017 000
main      10  suspend 0x00000078 0x00000800    16%   0x00000007 000
bus fault:
SCB_CFSR_BFSR:0x01 IBUSERR 
```

Lawlieta, 2019-04-23:
感觉像是你的netdev网卡设备没有成功注册添加一样，你确定一下 esp8266_netdev_add 函数调用是否成功，我这边也试着复现一下问题。

z14git, 2019-04-23:
@Lawlieta esp8266_netdev_add 函数有成功调用的。下图是进入hard fault前的状态：
[image]

Lawlieta, 2019-04-23:
你试一下把 `check_netdev_internet_up_work` 函数初始化时 sockfd 改为 -1 ，看还会不会出现 hard fault 问题。

[image]

z14git, 2019-04-23:
@Lawlieta 把sockfd初始化为-1后没有出现hard fault了👌

Lawlieta, 2019-04-23:
所以改完之后模块能正常运行了吗？ping功能能正常运行吗，ifconfig 查看一下当前网卡状态看看

z14git, 2019-04-24:
改完后这个问题解决了，ping,ifconfig都正常。

closed 2019-04-24 by z14git

Lawlieta, 2019-04-24:
这个问题已经在最新源码中修改过了，你可以更新到最新源码使用 https://github.com/RT-Thread/rt-thread/pull/2609