RT-Thread-packages/at_device #53: 更新at组件编译时发生错误

ruizhai1, 2019-05-27:
更新at组件编译时发生错误
[image]
D:\rt-thread\components\net\at\at_socket下的struct at_device_ops和D:\rt-thread\bsp\ls1cdev\packages\at_device-latest\inc下的struct at_device_ops重复
[image]
[image]

Lawlieta, 2019-05-27:
需要更新 RT-Thread 最新源码

Ghazigq, 2019-06-04:
> 
> 
> 需要更新 RT-Thread 最新源码，最新源码发布了 AT 组件 V1.3.0 版本

我的也是这种问题，用的是V2.0.0的AT组件，内核4.0.1

Lawlieta, 2019-06-04:
可以使用最新的 4.0.2 版本系统

Ghazigq, 2019-06-04:
> 
> 
> V2.0.0 的 AT device 需要使用最新的 4.0.2 版本系统哦
好的

closed 2019-09-02 by armink