RT-Thread-packages/at_device #84: mqtt sim800c 收到大量数据导致 接收失败

liaolj, 2019-12-01:
01-01 00:00:00 E/at.skt at_clnt: no memory for sim800c device(sim0) URC receive buffer (1348).
01-01 00:00:00 E/at.skt at_clnt: no memory for sim800c device(sim0) URC receive buffer (1460).
01-01 00:00:00 E/at.skt at_clnt: no memory for sim800c device(sim0) URC receive buffer (1460).
01-01 00:00:00 E/at.skt at_clnt: no memory for sim800c device(sim0) URC receive buffer (1124).
后面就一直不行了

liaolj, 2019-12-01:
我想可能at_device 所有驱动都有这个问题。马上项目要用E20 ，到时候测试下

Lawlieta, 2019-12-05:
接收数据以链表形式存储，接收数据过快会出现内存不足现象，底层 URC 机制不好做流控处理容易产生数据丢失的现象，建议在应用层分块接收数据，或者做延时接收处理，可以防止上述现象

liaolj, 2019-12-06:
我们是在模拟测试的时候发生过这种情况，设备端没法控制服务器端的控制逻辑的。

Lawlieta, 2019-12-06:
是否试过在应用层加延时接收处理，给数据处理提供足够的时间，看是否能解决问题

liaolj, 2019-12-08:
设备端的应用层加了没有用，因为是在驱动层，程序就不行了，恢复不了。只有在服务器端的应用层才有效。但是我们的设备是卖给别人的，如果他人的应用程序发大数据包，我们是没法控制，所有看看驱动这块有没有办法

armink, 2019-12-09:
> 如果他人的应用程序发大数据包，我们是没法控制，所有看看驱动这块有没有办法

试着给通讯接口增加硬件或软件流控，保证接收方不要超负荷

liaolj, 2019-12-10:
这是个办法，试试看

closed 2020-04-22 by Lawlieta