RT-Thread-packages/at_device #85: at_device 小包数据还是会丢失，还没有解决这个问题。看起来像

liaolj, 2019-12-05:
01-01 00:00:00 E/at.clnt at_clnt: AT Client receive failed, uart device get data error(-2)
01-01 00:00:00 E/at.skt.sim800 at_clnt: sim0 device receive size(2) data failed.
01-01 00:00:00 E/mqtt mqtt0: [1147090] wait Ping Response res: 0

Lawlieta, 2019-12-05:
能确定具体未接受到数据的原因吗，逻辑分析仪确定串口是否数据过来，中断有没有正常触发，at_client_recv 是否接收到数据，或者能否提供复现方式？

liaolj, 2019-12-06:
看日志是mqtt的呼吸包 就是2个字节  0x0d 0x0a

liaolj, 2019-12-06:
一定会复现的

majianjia, 2021-04-21:
我也遇到了，这个问题应该是用户应用关中断导致uart少收了几个字节。
我把critical内的代码改少后可以了。DMA的接收方式应该也可以。我用的LPUART1，没有DMA的支持。

majianjia, 2021-04-21:
我单步试了一下，有可能是ESP8266本身有问题，我的是V1.7.4的AT固件
收到的命令是 `+IPD0,4:`
但是串口只能拿出3个字节。

majianjia, 2021-04-21:
这是正常的，URC返回
~~~
[D/AT] recvline: 0000-0020: 53 45 4E 44 20 4F 4B 0D  0A                                                                           SEND OK..
[D/AT] sendline: 0000-0020: 41 54 2B 43 49 50 53 45  4E 44 3D 30 2C 32                                                            AT+CIPSEND=0,2
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] recvline: 0000-0020: 3E                                                                                                    >
[D/AT] sendline: 0000-0020: C0 00                                                                                                 ..
[D/AT] recvline: 0000-0020: 20 0D 0A                                                                                               ..
[D/AT] recvline: 0000-0020: 52 65 63 76 20 32 20 62  79 74 65 73 0D 0A                                                            Recv 2 bytes..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 53 45 4E 44 20 4F 4B 0D  0A                                                                           SEND OK..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 2B 49 50 44 2C 30 2C 32  3A                                                                           +IPD,0,2:
[D/AT] urc_recv: 0000-0020: D0 00                                                                                                 ..
[D/AT] sendline: 0000-0020: 41 54 2B 43 49 50 53 45  4E 44 3D 30 2C 31 30                                                         AT+CIPSEND=0,10
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
~~~

从这一刻起不正常
~~~
[D/AT] recvline: 0000-0020: 52 65 63 76 20 32 20 62  79 74 65 73 0D 0A                                                            Recv 2 bytes..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 53 45 4E 44 20 4F 4B 0D  0A                                                                           SEND OK..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 2B 49 50 44 2C 30 2C 32  3A                                                                           +IPD,0,2:
[D/AT] sendline: 0000-0020: 41 54 2B 43 49 50 53 45  4E 44 3D 30 2C 32                                                            AT+CIPSEND=0,2
[D/AT] recvline: 0000-0020: 00 0D 0A                                                                                              ...
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] recvline: 0000-0020: 3E                                                                                                    >
[D/AT] sendline: 0000-0020: C0 00                                                                                                 ..
[D/AT] recvline: 0000-0020: 20 0D 0A                                                                                               ..
[D/AT] recvline: 0000-0020: 52 65 63 76 20 32 20 62  79 74 65 73 0D 0A                                                            Recv 2 bytes..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 53 45 4E 44 20 4F 4B 0D  0A                                                                           SEND OK..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 2B 49 50 44 2C 30 2C 32  3A                                                                           +IPD,0,2:
[D/AT] sendline: 0000-0020: 41 54 2B 43 49 50 53 45  4E 44 3D 30 2C 32                                                            AT+CIPSEND=0,2
[D/AT] recvline: 0000-0020: 00 0D 0A                                                                                              ...
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] recvline: 0000-0020: 3E                                                                                                    >
~~~

可以看出这里ESP一直在发+IDP说有数据，但是一直没提取数据。
我觉得原因可能在收到+IDP后，提取数据时有一个等到信号量的过程。
这个过程里面被这个抢了`AT+CIPSEND=0,2`

mysterywolf, 2022-01-16:
https://club.rt-thread.org/ask/question/434391.html
看一下这个

mysterywolf, 2022-01-16:
https://github.com/RT-Thread/rt-thread/issues/3600

ztlchina, 2022-01-21:
> https://club.rt-thread.org/ask/question/434391.html 看一下这个

这个是我写的答案， 现在http-ota 使用 eg25 atclient 下载固件， 校验失败。 看来问题的确存在啊。 下载第一包1024个字节的时候，丢了两个。 再查查是咋回事。

ztlchina, 2022-01-21:
现在的情况汇报一下：
1）软硬件环境：STM32F407，EG25 ,  ATCLIENT, HTTP_OTA ， SPIFLASH, W25Q128;
现象：  使用串口中断接收模式， http_ota 下载完成重启后，download分区CRC 校验失败。
分析： 1） 定位问题到下载的数据不正确， 而不是flash内的数据写错误；
            2）尝试将串口缓存搞大点， （ESP8266的时候，底层串口缓存太小造成丢数据）因为http_ota下载一包数据是1024，加上几百个http头， 所以索性改成了2K的串口缓存。  试验结果不理想。 1024个字节，在中间偶尔丢1到两个， 然后将结尾的0x0d 0x0a 补进来。 
          3）尝试减小或者增加http_ota , at client 等各处的缓存。 无果。
          4） 最终尝试使用DMA方式， 解决。 CRC 通过。
---------------------------------------------------------------------------------------------------------
总结：
        EG25，等在中断串口模式下载大数据的时候， 会丢数， 即使将串口缓存开大也会丢。 改成DMA模式就OK了。 
另外http_ota 里有一个bug， 已经另外提交了issue。

majianjia, 2022-01-21:
是的，我感觉有可能是串口忙不过来。跟我之前遇到的情况一模一样，会偶尔丢一个字节。
你试试波特率降低，中断下应该会好转。