RT-Thread-packages/at_device #88: 在env中配置了esp8266后，在rt_config中生成的宏是AT_DEVICE_ESP8266，而SConscript脚本中宏条件都是AT_DEVICE_USING_ESP8266

lizhen9880, 2019-12-20:
在env中配置了esp8266后，在rt_config中生成的宏是AT_DEVICE_ESP8266，而SConscript脚本中天条件都是AT_DEVICE_USING_ESP8266，导致生成的工程中无at_device_esp8266.c等esp8266相关文件

Lawlieta, 2019-12-21:
你需要在 ENV 工具中使用 pkgs --upgrade 命令更新一下软件包索引信息

lizhen9880, 2019-12-23:
> 你需要在 ENV 工具中使用 pkgs --upgrade 命令更新一下软件包索引信息

更新过了，不是更新的问题，pkgs --upgrade 然后scons --target=mdk5，然后rt_config.h中开启ESP8266的宏是
`#define AT_DEVICE_ESP8266`
而SConscript和.config中是
`if GetDepend(['AT_DEVICE_USING_ESP8266']):`
导致8266的相关文件无法添加到src中

Lawlieta, 2019-12-23:
首先问一下你使用什么版本软件包，什么版本RT-Thread系统？
然后我估计你是 ENV 工具软件包索引没有更新成功，一直保持的之前的选项， 之前宏定义是 AT_DEVICE_XXX 宏定义后面统一改成 AT_DEVICE_USING_XXX 宏定义，你可以看一下你 ENV 工具对应目录路径是否是如下图所示：

[image]

如果不是，直接点的方法，可以删除当前 ENV 路径，在官网下载最新的 ENV 软件包，再执行 pkgs --upgrade 看一下有没有正常

lizhen9880, 2019-12-29:
> 首先问一下你使用什么版本软件包，什么版本RT-Thread系统？
> 然后我估计你是 ENV 工具软件包索引没有更新成功，一直保持的之前的选项， 之前宏定义是 AT_DEVICE_XXX 宏定义后面统一改成 AT_DEVICE_USING_XXX 宏定义，你可以看一下你 ENV 工具对应目录路径是否是如下图所示：
> 
> [image]
> 
> [image]
> 
> 如果不是，直接点的方法，可以删除当前 ENV 路径，在官网下载最新的 ENV 软件包，再执行 pkgs --upgrade 看一下有没有正常

按照您的描述做的，问题解决了，谢谢

closed 2019-12-29 by lizhen9880