RT-Thread-packages/at_device #97: RTT studio开启at_device组件后编译报错

ReCclay, 2020-02-29:
请问出现下面报错信息，该如何解决呢？

```
arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -O0 -ffunction-sections -fdata-sections -Wall  -g -gdwarf-2 -T "D:\RT-ThreadStudio\workspace\clock_v2\link.lds" -Xlinker --gc-sections -Wl,-Map,"rtthread.map" -Xlinker --cref --specs=nano.specs -o "rtthread.elf"  ./rt-thread/src/clock.o ./rt-thread/src/components.o ./rt-thread/src/device.o ./rt-thread/src/idle.o ./rt-thread/src/ipc.o ./rt-thread/src/irq.o ./rt-thread/src/kservice.o ./rt-thread/src/mem.o ./rt-thread/src/mempool.o ./rt-thread/src/object.o ./rt-thread/src/scheduler.o ./rt-thread/src/signal.o ./rt-thread/src/thread.o ./rt-thread/src/timer.o  ./rt-thread/libcpu/arm/cortex-m3/context_gcc.o ./rt-thread/libcpu/arm/cortex-m3/cpuport.o  ./rt-thread/libcpu/arm/common/backtrace.o ./rt-thread/libcpu/arm/common/div0.o ./rt-thread/libcpu/arm/common/showmem.o  ./rt-thread/components/net/sal_socket/src/sal_socket.o  ./rt-thread/components/net/sal_socket/socket/net_netdb.o  ./rt-thread/components/net/sal_socket/impl/af_inet_at.o  ./rt-thread/components/net/netdev/src/netdev.o ./rt-thread/components/net/netdev/src/netdev_ipaddr.o  ./rt-thread/components/net/at/src/at_cli.o ./rt-thread/components/net/at/src/at_client.o ./rt-thread/components/net/at/src/at_utils.o  ./rt-thread/components/net/at/at_socket/at_socket.o  ./rt-thread/components/libc/compilers/newlib/libc.o ./rt-thread/components/libc/compilers/newlib/libc_syms.o ./rt-thread/components/libc/compilers/newlib/stdio.o ./rt-thread/components/libc/compilers/newlib/syscalls.o ./rt-thread/components/libc/compilers/newlib/time.o  ./rt-thread/components/finsh/cmd.o ./rt-thread/components/finsh/msh.o ./rt-thread/components/finsh/shell.o  ./rt-thread/components/drivers/src/completion.o ./rt-thread/components/drivers/src/dataqueue.o ./rt-thread/components/drivers/src/pipe.o ./rt-thread/components/drivers/src/ringblk_buf.o ./rt-thread/components/drivers/src/ringbuffer.o ./rt-thread/components/drivers/src/waitqueue.o ./rt-thread/components/drivers/src/workqueue.o  ./rt-thread/components/drivers/serial/serial.o  ./rt-thread/components/drivers/sensors/sensor.o ./rt-thread/components/drivers/sensors/sensor_cmd.o  ./rt-thread/components/drivers/misc/pin.o  ./rt-thread/components/drivers/i2c/i2c-bit-ops.o ./rt-thread/components/drivers/i2c/i2c_core.o ./rt-thread/components/drivers/i2c/i2c_dev.o  ./packages/dht11-latest/dht11_sample.o ./packages/dht11-latest/sensor_dallas_dht11.o  ./packages/at_device-v2.0.1/src/at_device.o  ./packages/at_device-v2.0.1/samples/at_sample_esp8266.o  ./packages/at_device-v2.0.1/class/esp8266/at_device_esp8266.o ./packages/at_device-v2.0.1/class/esp8266/at_socket_esp8266.o  ./libraries/STM32F1xx_HAL_Driver/Src/Legacy/stm32f1xx_hal_can.o  ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_adc.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_adc_ex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_can.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_cec.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_cortex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_crc.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_dac.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_dac_ex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_dma.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_eth.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_exti.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash_ex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_gpio.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_gpio_ex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_hcd.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_i2c.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_i2s.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_irda.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_iwdg.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_mmc.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_nand.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_nor.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pccard.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pcd.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pcd_ex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pwr.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rcc.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rcc_ex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rtc.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rtc_ex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_sd.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_smartcard.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_spi.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_sram.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_uart.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_usart.o ./libraries/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_wwdg.o  ./libraries/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/startup_stm32f103xb.o  ./libraries/CMSIS/Device/ST/STM32F1xx/Source/Templates/system_stm32f1xx.o  ./drivers/board.o ./drivers/drv_common.o ./drivers/drv_gpio.o ./drivers/drv_soft_i2c.o ./drivers/drv_usart.o  ./applications/main.o   
d:/rt-threadstudio/platform/env_released/env/tools/gnu_gcc/arm_gcc/mingw/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: rtthread.elf section `.text' will not fit in region `ROM'
d:/rt-threadstudio/platform/env_released/env/tools/gnu_gcc/arm_gcc/mingw/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: region `ROM' overflowed by 42708 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:72: rtthread.elf] Error 1
"make -j4 all" terminated with exit code 2. Build might be incomplete.

15:59:06 Build Failed. 3 errors, 2 warnings. (took 23s.999ms)

```

armink, 2020-02-29:
region `ROM' overflowed by 42708 bytes

ReCclay, 2020-02-29:
我用的是STM32F103C8T6是这个单片机的ROM不够吗？

armink, 2020-02-29:
是的，要做些裁剪

ReCclay, 2020-02-29:
可是我只用到了8266，其他东西好像并没有参与编译嘞，请教该如何裁剪呢？

Lawlieta, 2020-04-22:
STM32F103C8T6 64K ROM 很难上 AT Socket 功能，建议直接使用 AT Client 功能

closed 2020-04-22 by Lawlieta