RT-Thread-packages/at_device #106: ESP8266在使用中会出现崩溃的情况

akinggw, 2020-04-16:
在at_socket_esp8266.c文件中，进行跟踪，每次正常连接后，就出现崩溃，通过跟踪，发现出现的崩溃的地方在下面这个函数：
    
/* parse response data, get the DHCP status */
    if (at_resp_parse_line_args_by_kw(resp, "+CWDHCP_CUR:", "+CWDHCP_CUR:%d", &dhcp_stat) < 0)
    {
        LOG_E("%s device prase DHCP status error.", device->name);
        goto __exit;
    }

这里这么写感觉有点问题：
 "+CWDHCP_CUR:%d", &dhcp_stat)

dhcp_stat的定义：

rt_uint8_t dhcp_stat = 0;

改成
    if (at_resp_parse_line_args_by_kw(resp, "+CWDHCP_CUR:", "+CWDHCP_CUR:%d", dhcp_stat) < 0)
后一切正常。

Lawlieta, 2020-04-16:
&dhcp_stat 没问题，你可以看一下 at_resp_parse_line_args_by_kw 函数底层实现，底层调用的是 vsscanf 函数。

问题应该出在 dhcp_stat 变量的类型上，现在是 uint8_t 类型，vsscanf 对入参类型检查比较严格，%d 来解析 uint8_t 类型参数，会出现越界的问题，可能会导致你现在错误。

你试着将 dhcp_stat 改为 int 型或者 uint32_t 类型，应该也没问题，感谢问题反馈，也欢迎提交 PR 修复该问题。

Lawlieta, 2020-04-22:
已修改 https://github.com/RT-Thread-packages/at_device/pull/108

closed 2020-04-22 by Lawlieta