RT-Thread-packages/at_device #112: ME3616数据收发不完整

LXGMAX, 2020-05-12:
[image]
尝试AT_CMD_MAX_LEN 修改到512，未果；
尝试ME3616_SAMPLE_RECV_BUFF_LEN 修改到1024，未果。
使用STM32F103RCT6 UART2

Lawlieta, 2020-05-13:
从日志看是模块响应的数据有问题（发送 AT 返回 K\r\n，正常应返回 OK\r\n），导致错误提示，确定一下是否是模块接线问题或者 AT 模块是否正常启动

LXGMAX, 2020-05-13:
> 从日志看是模块响应的数据有问题（发送 AT 返回 K\r\n，正常应返回 OK\r\n），导致错误提示，确定一下是否是模块接线问题或者 AT 模块是否正常启动

我看官方手册是<CR><LF>OK<CR><LF>响应的，感觉是这个的问题？

me3616直接接串口是能正常通信和控制的。

Lawlieta, 2020-05-13:
> 
> 
> > 从日志看是模块响应的数据有问题（发送 AT 返回 K\r\n，正常应返回 OK\r\n），导致错误提示，确定一下是否是模块接线问题或者 AT 模块是否正常启动
> 
> 我看官方手册是“OK“响应的，感觉是这个的问题？
> 
> me3616直接接串口是能正常通信和控制的。

现在接收的数据格式不对，每次接收固定前面几个字节丢失了，感觉像串口驱动的问题，你用逻辑分析仪抓一下串口数据看看

LXGMAX, 2020-05-13:
> > 
> 
> > 
> 
> > > 从日志看是模块响应的数据有问题（发送 AT 返回 K\r\n，正常应返回 OK\r\n），导致错误提示，确定一下是否是模块接线问题或者 AT 模块是否正常启动
> 
> > 
> 
> > 我看官方手册是“OK“响应的，感觉是这个的问题？
> 
> > 
> 
> > me3616直接接串口是能正常通信和控制的。
> 
> 
> 
> 现在接收的数据格式不对，每次接收固定前面几个字节丢失了，感觉像串口驱动的问题，你用逻辑分析仪抓一下串口数据看看

[image]
感觉是这个导致部分数据丢失，不符合at组件的响应处理方式

Lawlieta, 2020-05-13:
@qiyongzhong0 齐工，你有时间确定一下 ME3616 这个问题吧

qiyongzhong0, 2020-05-15:
@LXGMAX 你看下UART2被几个地方使用了

qiyongzhong0, 2020-05-15:
### @LXGMAX 正确的开机输出信息如下：

```
[2020-05-15_09:26:17:511]
[2020-05-15_09:26:17:511]*MATREADY: 1

[2020-05-15_09:26:17:511]+CFUN: 1
[2020-05-15_09:26:18:227]
[2020-05-15_09:26:18:227]+CPIN: READY
```

### Hex格式如下：

```
[2020-05-15_09:26:17:511]0D 0A 
[2020-05-15_09:26:17:511]2A 4D 41 54 52 45 41 44 59 3A 20 31 0D 0A 0D 0A 
[2020-05-15_09:26:17:511]2B 43 46 55 4E 3A 20 31 0D 0A 
[2020-05-15_09:26:18:227]0D 0A 
[2020-05-15_09:26:18:227]2B 43 50 49 4E 3A 20 52 45 41 44 59 0D 0A
``` 

### 你的输出信息中，不只是开头少几个字节，中间也少字节。

qiyongzhong0, 2020-05-15:
### @LXGMAX 系统主频设置多少，如果主频太低可能导致串口接收数据处理不及时出现丢数据问题，提高主频试试。

LXGMAX, 2020-05-16:
> ### @LXGMAX 系统主频设置多少，如果主频太低可能导致串口接收数据处理不及时出现丢数据问题，提高主频试试。

已经解决，是时钟的问题，感谢！

closed 2020-05-16 by LXGMAX