RT-Thread-packages/at_device #120: bc26无法正常工作

jch12138, 2020-05-27:
bc26在初始化时,会卡在获取dns这一步,我尝试手动发送"AT+QIDNSCFG=1"指令,经常会无法收到返回结果,或是要等很久才会返回.
我在[这里](https://www.rt-thread.org/qa/forum.php?mod=viewthread&tid=423606&highlight=bc26)看到了一样的情况,网友说是bc26不需要向运营商获取dns,内部自带dns,可以帮忙解决一下吗@qiyongzhong0 
[image]

jch12138, 2020-05-27:
[image]

qiyongzhong0, 2020-05-27:
@jch12138 
手动命令有回OK
代码中,获取DNS配置命令的超时时间为300ms,不可能卡住的.
```
static int bc26_netdev_set_info(struct netdev *netdev)
{
#define BC26_INFO_RESP_SIZE 128
#define BC26_INFO_RESP_TIMO rt_tick_from_millisecond(300)
```
```
/* send "AT+QIDNSCFG=1" commond to get DNS servers address */
        if (at_obj_exec_cmd(device->client, resp, "AT+QIDNSCFG=1") != RT_EOK)
        {
            result = -RT_ERROR;
            goto __exit;
        }

        /* parse response data "+QIDNSCFG: <contextID>,<pridnsaddr>,<secdnsaddr>" */
        if (at_resp_parse_line_args_by_kw(resp, "+QIDNSCFG:", "+QIDNSCFG: 1,\"%[^\"]\",\"%[^\"]\"",
                                          dns_server1, dns_server2) <= 0)
        {
            LOG_E("%s device prase \"AT+QIDNSCFG=1\" cmd error.", device->name);
            result = -RT_ERROR;
            goto __exit;
        }
```

jch12138, 2020-05-27:
> @jch12138
> 手动命令有回OK
> 代码中,获取DNS配置命令的超时时间为300ms,不可能卡住的.
>
> ```
> static int bc26_netdev_set_info(struct netdev *netdev)
> {
> #define BC26_INFO_RESP_SIZE 128
> #define BC26_INFO_RESP_TIMO [...]
可能是因为我模块或者是基站的问题,我手动查询,返回ok后,可能要好几分钟才返回dns信息,我们实验室的bc26模块都不太正常,目前把域名直接替换为ip地址,可以正常工作,暂时解决问题

closed 2022-01-12 by mysterywolf