RT-Thread-packages/at_device #153: ec200x 不使用电源引脚时无法检测网络状态

huqifj, 2021-01-18:
您好，感谢 **at_device**。

我在使用 ec200t 时遇到一个问题，模块 at 命令初始化之后一直打印 **the power is off.**，间隔为 `EC200X_LINK_DELAY_TIME`。

```C
static int ec200x_check_link_status(struct at_device *device)
{
    at_response_t resp = RT_NULL;
    struct at_device_ec200x *ec200x = RT_NULL;
    int result = -RT_ERROR;
    
    ec200x = (struct at_device_ec200x *)device->user_data;
    if ( ! ec200x->power_status)//power off
    {
        LOG_D("the power is off.");
        return(-RT_ERROR);
    }
   /* ... */
}
```

目前我是做了以下修改规避这个问题
```bash
--- a/class/ec200x/at_device_ec200x.c
+++ b/class/ec200x/at_device_ec200x.c
@@ -44,6 +44,7 @@ static int ec200x_power_on(struct at_device *device)

     if (ec200x->power_pin == -1)//no power on pin
     {
+        ec200x->power_status = RT_TRUE;
         return(RT_EOK);
     }
     if (ec200x->power_status_pin != -1)//use power status pin
@@ -82,6 +83,7 @@ static int ec200x_power_off(struct at_device *device)

     if (ec200x->power_pin == -1)//no power on pin
     {
+        ec200x->power_status = RT_FALSE;
         return(RT_EOK);
     }
     if (ec200x->power_status_pin != -1)//use power status pin
```

不知道这是 bug 还是我使用方式不对。

xiangxistu, 2021-01-18:
这个应该是一个设计思路的问题，power_status 作为使用依据了。

huqifj, 2021-01-18:
使用电源控制引脚 `EC200X_SAMPLE_POWER_PIN` 的时候，`power_status` 作为使用依据没问题，

但配置 `#define EC200X_SAMPLE_POWER_PIN -1` 的时候，某些代码依然判断了 `power_status` 的值，貌似没有一个 API 让用户去设置 `power_status` 的值