RT-Thread-packages/at_device #189: AT_USING_SOCKET_SERVER  与 const 的冲突。解决#182引起的

ztlchina, 2022-11-16:
最新版本（2022-11-16 19:28:24） 开启了宏定义AT_USING_SOCKET_SERVER 与 atclient 包里的     const struct at_socket_ops *socket_ops;      /* AT device socket operations */
冲突了。 

#182 
解决了182. 引起了新的问题。

按下葫芦浮起瓢。。。

ztlchina, 2022-11-16:
编译错误：
../../../components/net/at/at_socket/at_socket.c(771): error: cannot assign to non-static data member 'socket_ops' with const-qualified type 'const struct at_socket_ops *'

ztlchina, 2022-11-16:
at_device.h:
struct at_device_class
{
    uint16_t class_id;                           /* AT device class ID */
    const struct at_device_ops *device_ops;      /* AT device operaiotns */
#ifdef AT_USING_SOCKET
    uint32_t socket_num;                         /* The maximum number of sockets support */
    const struct at_socket_ops *socket_ops;      /* AT device socket operations */
#endif
    rt_slist_t list;                             /* AT device class list */
};

```

    /* avoid use bottom driver to alloc "socket" */
    store_at_socket_temporary = device->class->socket_ops->at_socket;
    device->class->socket_ops->at_socket = RT_NULL;
    new_sock = alloc_socket_by_device(device, AT_SOCKET_TCP);
    if (new_sock == RT_NULL)
    {
        return;
    }
    new_sock->type = AT_SOCKET_TCP;
    new_sock->state = AT_SOCKET_CONNECT;

    /* set AT socket receive data callback function */
    new_sock->ops->at_set_event_cb(AT_SOCKET_EVT_RECV, at_recv_notice_cb);
    new_sock->ops->at_set_event_cb(AT_SOCKET_EVT_CLOSED, at_closed_notice_cb);
    new_sock->ops->at_set_event_cb(AT_SOCKET_EVT_CONNECTED, at_connect_notice_cb);
    device->class->socket_ops->at_socket = store_at_socket_temporary;
```

ccx-lan, 2023-11-07:
我直接把const去掉
[image]

yangpengya, 2023-12-26:
直接把store_at_socket_temporary删了，没用